linien-org/linien #118: linien server installation issues

aisichenko, 2020-11-07:
I have tried to install linien (both 0.2.3 and 0.2.2) onto several Red Pitayas and each time after I select "Yes" to installing the linien server on red pitaya, packages are loaded and then half-way through the installation, the linien app freezes and the Red Pitaya orange LED indicating access to SD card stops flashing. After this I usually reinstall the Red Pitaya OS onto the SD card. 

Each time I try reinstalling the linien server, the freezing occurs at different points of the installation.

I have tried this with the Red Pitaya releases 0.99-48 and 1.03-701 on two different Red Pitayas.

Can I load the required linien server onto the Red Pitaya SD directly? How do I do this?

hermitdemschoenenleben, 2020-11-07:
Hi! I had a similar problem once, and it was caused by a faulty SD card that stopped working once a specific cell of the card was addressed. The orange LED indicator you describe also points in this direction. It is strange though that the problem occurs with two RedPitayas independently. Did you use the same SD card in both cases?

I'd suggest that you test whether your card(s) are really broken by writing data to it until it is completely full. Additionally: do you have a new SD card you could use for trying to install linien?

> Can I load the required linien server onto the Red Pitaya SD directly? How do I do this?
at the moment this is not possible. If your problem persists, I can check whether I would be allowed to distribute a modified RedPitaya OS with Linien preinstalled. Actually, this may be really a good idea for easier installation, thank you :)

aisichenko, 2020-11-13:
This issue was solved by trying different SD cards and making sure the ethernet connection was proper. Whatever it was, it is a combination of loading files from the internet onto the RP and installing them to the SD that ran into some bottleneck. For users that see this issue in the future, I suggest (but did not have time to fully prove why) using 16 GB SD card and connecting the RP to a router that you know works. Also, trying the linien server install several times - each time it fails, reflash the SD card with 0.99-48 OS, and try reinstalling linien again.

closed 2020-11-13 by aisichenko