linien-org/linien #124: Monitoring the spectroscopy signal during lock

aisichenko, 2020-11-13:
Not sure if this is a option already, but having a running monitor on the GUI of the spectroscopy signal (IN1) would be useful, similar to the "control history".

Also, a real time monitor of the error signal's power spectral density would be great. Something like this:

```python
from scipy import signal    
def residual_freq_noise(self, times, sig):
        """ Calculates the residual frequency noise.
        Calculate PSD using welch modified periodogram with hann window

        Args:
            times: array of times
            sig: signal array, e.g. error signal

        Returns:
            f: offset frequencies
            psd: residual frequency noise PSD in units V^2/Hz

        """
        dt = times[1] - times[0]
        fs = 1 / dt

        num_pts = int(len(sig))

        hann = signal.hann(num_pts)

        f, psd = signal.welch(sig, fs=fs, window=hann)

        return f, psd
```

hermitdemschoenenleben, 2020-11-13:
Thank you for the ideas! An analysis of the error signal's PSD would be really nice, and I plan to implement it. Regarding your first suggestion (monitor of spectroscopy signal) I'm not quite sure if I understand the idea correctly. Is it the DC signal that you'd like to monitor? Or are you talking about the spectroscopy signal when the laser is locked?

aisichenko, 2020-11-13:
Yes that's right - monitoring the DC spectroscopy signal when the laser is locked (monitoring IN1)