linien-org/linien #133: Ask some questions about linien

Yao-Chin, 2020-12-07:
Hi guy,

The linien software is well and more help my experiment.
But I met some questions, I don't know how to fix from software code. 

First, about PID value question. I do some testing. I know that the unity gain is P value (Proportional gain) of 1024. Low frequency range will be integrate when I value (Integral gain) turn on.     
However, nothing happened when the D value (Differential gain) be changed. And then, how to know the transfer function from these values (P, I and D)?

Second, about glitch noise issue.
I use linien application to feedback control system. 
I also monitor error signal (green line on ch1) and laser transmission light (pink line on ch4) shown on oscilloscope, but i find some glitch noise as below figure (link).
https://drive.google.com/file/d/1sQ3QNfYUXOS01zp4LNOn1G95r9qJU3Rp/view?usp=sharing
If I remove ethernet cable between Red Pitaya and PC, the noise will disappear.
But I want to continuity receive data of Red Pitaya through Ethernet cable.
Do you good ideal to solve? 

Thank you very much
Best regards,
Yao-Chin Huang

hermitdemschoenenleben, 2020-12-30:
Hello,

yes, in the previous versions the derivative was indeed broken. I just fixed this issue though and uploaded version 0.3.1.
In this version, unity P gain is 4096. The transfer function is
`
L(f) = kp + ki / f + kd * f
`
with `kp=P/4096`, `ki=I/0.1s` and `kd=D / (2**6 * 125e6)`.
In the next days, I'll also update the documentation with this information.
If you want to plot the transfer function, have a look at `tests/test_pid_transfer.py` which simulates the PID code that runs on FPGA (`plot_transfer()`). There is also `plot_theory()` which plots the theoretical transfer function.
There are two caveats though: This transfer function does not include filtering stages that run between demodulation and PID, and the frequency response of the RedPitaya hardware is not flat as well.

About the glitch noise issue: Thank you very much for the report. We were able to reproduce this problem in the lab. I'm not quite sure yet, but this problem seems to be caused by the RedPitaya hardware itself, so probably the only thing that can be done about it is unplugging the ethernet cable. I will look into this issue within the next few days, maybe there is a better way to fix it.

hermitdemschoenenleben, 2021-03-03:
@Yao-Chin a colleage has found this:
https://ln1985blog.wordpress.com/2016/02/07/red-pitaya-dac-performance/
Apparently there is a crosstalk between digital and analog outputs of RedPitaya, which may be related to your problem. There's an easy solution for the problem (de-solder some resistors) which I'll try out in the next days

Yao-Chin, 2021-03-18:
@hermitdemschoenenleben 
Thank you for great suggestion. 
I also try it following blog's process.

By the way, recently, I found some report that "Ethernet LED blinking changes DAC voltage" issue as below link.
https://github.com/RedPitaya/RedPitaya/issues/205

Following previous result, the noise disappeared on removing the ethernet port condition.
It may be root cause.  
Did you have other suggestion to improve it?
Later, I will try to solve it following that web solution.

Thanks for your help.

hermitdemschoenenleben, 2021-03-23:
Thank you for this very helpful link! I just built in a fix that disables ethernet LED blinking when linien starts and enables it again when linien shuts down.

closed 2021-06-15 by hermitdemschoenenleben