linien-org/linien #145: Add support for rpyc 5.0.0

bleykauf, 2021-01-11:
Currently version ==4.1.5 is required. 5.0.0 is not backwards compatible.

hermitdemschoenenleben, 2021-01-11:
unfortunately right now this isn't possible as rpyc 5 requires python 3.6 and the RedPitaya ecosystem is still at 3.5. I'll keep this open though, hoping that RP base image will get an update

qku, 2023-09-25:
I would highly appreciate support for rpyc 5.0 as I would like to use linien together with a package already using it.

Let me know if you need help with e. g. testing or so.

bleykauf, 2023-10-16:
@qku It is very much possible that simply upgrading the version in the `setup.py` of the server and the client and then manually installing these custom version following [these instructions](https://github.com/linien-org/linien/wiki/Development).

qku, 2023-11-23:
So what works for us is to checkout the `feature/use-rpyc-5` branch on the client and install it, while the server runs normal 0.8.0 (with rpyc < 5.0).

As far as I can tell from my limited experience with linien, everything is working as expected. We are only using linien for a very simple PID setup though, without GUI but interfacing via python from another program.

qku, 2024-02-09:
Would more testing help you with the upgrade to rpyc 5? We would appreciate an upgrade to facilitate the use of linen together with other packages using rpyc.

PR #358 (2023-11-30): Use rpyc 5
Closes #145
PR #390 (2024-04-08): Release v2.0.0
### Added
* Use features of Python 3.10 available on RedPitaya OS 2.0 for `linien-server` by @bleykauf in https://github.com/linien-org/linien/pull/366
* Add ability to start the server upon startup by @bleykauf in https://github.com/linien-org/linien/pull/387

### Changed
* Use systemd instead of screen for running the server by @bleykauf in https://github.com/linien-org/linien/pull/387 (thanks @doronbehar  for the idea!)
* Use json to store devices and parameters by @bleykauf in https://github.com/linien-org/linien/pull/357
* Better error handling by @bleykauf in https://githu [... cut]
closed 2024-04-08 by bleykauf