linien-org/linien #152: GUI ramp control amplitude

aisichenko, 2021-02-03:
Hello, I've searched but could not find it: is there a control of the ramp amplitude? I see that there is a ramp frequency input. It would be very useful to add amplitude control. Thanks!

hermitdemschoenenleben, 2021-02-03:
@aisichenko ramp amplitude is controlled via "zoom" in the top left of the window. I should probably rename this control in order to state more clearly that this actually controls ramp amplitude and not only the data that is displayed

aisichenko, 2021-02-03:
I see the ramp amplitude changing, thanks! And yes, I agree, that would make it more clear. Perhaps two options: one that is just zoom, another that is "enable ramp manual control" that will have options for ramp freq, ramp amplitude, and ramp offset. Thanks again

closed 2021-02-17 by hermitdemschoenenleben