linien-org/linien #153: Ability to change "Watch Lock" while lock is running

bleykauf, 2021-02-17:
For example to prevent autolock from starting when adjusting laser current manually. 

Currently the check boxes are not visible when the lock is running. Threshold should also be adjustable.

hermitdemschoenenleben, 2021-03-03:
* [ ] Ability to change "Watch Lock" while lock is running
* [ ] Use DC spectroscopy signal

bleykauf, 2021-03-16:
Very simple way to see if lock is lost is to check whether the control signal rails at its maximum (maybe making this user configurable). Only catches some cases but is a very simple indicator.

Another potential issue to maybe keep in mind: There is also the case where there is no DC signal, i.e. when directly feeding in an error signal (offset locks) without demodulation. Also, the offset frequency might be changed in some setups (frequency jumps), i.e. a rapid change in the control signal is expected and is also not a good indicator for loss-of-lock. Specifically for this case a railing control signal (or non-zero error signal over an extended time) might be a good indicator.

bleykauf, 2021-07-23:
Some other suggestions/ideas:

* In the case that the output rails it might be a good to change the status from "locked" to something else. Useful for offset locks with an "infinite" capture range.
* Initiating a relock from a TTL signal might also be useful
* Add user configurable "cooldown" time before attempting relock, maybe maximum number off attempts