linien-org/linien #158: Ramp signal display information

aisichenko, 2021-03-03:
Hello,

Thanks for your great work! I was wondering if we can include a display of the ramp status. Specifically the ramp amplitude or Vpp and the ramp DC offset. 

This is useful for locking using an AOM system, where we'd like to make sure that we are not applying too much voltage to our VCO control port.

Thanks!

hermitdemschoenenleben, 2021-03-03:
Hi,

a few days ago I pushed this to the master branch:
[image]
I'll try to release a new version of Linien next week including this change, I'll notify you then

hermitdemschoenenleben, 2021-03-04:
@aisichenko one thing you should keep in mind: when starting Linien server on RedPitaya, the FPGA image is re-flashed. Then, there is a short period of time before the client has set all parameters, i.e. the output may be in an undefined state for a second or so --> it's best to wait until everything is completely running before connecting your sensitive device

aisichenko, 2021-03-04:
Understood. Thanks!!

hermitdemschoenenleben, 2021-03-10:
@aisichenko I just uploaded 0.4.0 (and the bugfix release 0.4.1) which includes the ramp signal display. I also implemented recording PSD of error signal (#124) when the laser is locked - if you have any thoughts on this, your comments are more than welcome!

closed 2021-03-10 by hermitdemschoenenleben