linien-org/linien #184: ModuleNotFoundError when trying to start (fake) server from command line

bleykauf, 2021-09-02:
According to the readme, I should be able to start a fake server with `python3 server/server.py --fake`.

However, this results in import errors:
```python
Traceback (most recent call last):
  File "server\server.py", line 17, in <module>
    from autolock.autolock import Autolock
  File "C:\Users\bastian\Documents\programming\linien\linien\server\autolock\autolock.py", line 6, in <module>
    from linien.common import (
ModuleNotFoundError: No module named 'linien.common'
```

Note that I din't install linien in this environment so `linien` is not accessible from `server.py`.

If install linien via pip, The error changes to  

```python
Traceback (most recent call last):
  File "server\server.py", line 17, in <module>
    from autolock.autolock import Autolock
  File "C:\Users\bastian\Documents\programming\linien\linien\server\autolock\autolock.py", line 12, in <module>
    from linien.server.autolock.algorithm_selection import AutolockAlgorithmSelector
ModuleNotFoundError: No module named 'linien.server'
```

I believe this is since `linien/server` itself is not a module (there is no `__init__.py` inside the folder). But why does `import linien.server` not result in an error then?

I could resolve some of the relative import errors by replacing `linien.` with `..`:

```python
from .autolock.autolock import Autolock
from .parameters import Parameters

from ..config import DEFAULT_SERVER_PORT
from ..common import (
    check_plot_data,
    update_signal_history,
    N_POINTS,
    pack,
    unpack,
)
from ..server.parameter_store import ParameterStore
from ..server.optimization.optimization import OptimizeSpectroscopy
from ..server.pid_optimization.pid_optimization import (
    PIDOptimization,
    PSDAcquisition,
```

However, relative imports don't work when running `server.py` as a script. From within python, this works without error `from linien.server.server import run_server`. However, I also cannot start this, due to the `@click` decorator:

```python
>>> run_server(18862, True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\bastian\miniconda3\envs\linien_pyinstall_py36\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\bastian\miniconda3\envs\linien_pyinstall_py36\lib\site-packages\click\core.py", line 767, in main
    args = list(args)
TypeError: 'int' object is not iterable
```

Wouldn't the current situtation also import everything from the pip installed version when running `linien_start_server` rather then the development version when running on a RP?

hermitdemschoenenleben, 2021-09-02:
> According to the readme, I should be able to start a fake server with python3 server/server.py --fake.

You're right, running the server like you tried does not work. Sorry for that! For me, running it from the parent folder works though:
```
python3 linien/server/server.py --fake
```

In this case, the working directory contains the linien folder, i.e. imports from `linien.XXX` are found.

> I believe this is since linien/server itself is not a module 

I'm quite surprised that the server folder doesn't contain a `__init__.py`. And I honestly don't know why imports like `from linien.server.XXX` do work at all. As I understand the python import system it shouldn't... So yeah, the folder should countain an init file.

> Wouldn't the current situtation also import everything from the pip installed version when running linien_start_server rather then the development version when running on a RP?

exactly. With the current way the imports work, it is required to uninstall the pip version on red pitaya in order to use the dev version. A while back I tried changing all imports to relative ones, but this resulted in mysterious errors that occured when building / launching the pyinstaller bundle. I was not able to resolve these, so I left the absolute imports unchanged

bleykauf, 2021-09-03:
Just tried again starting the server from the parent directory and it still didn't work for me.

```
Traceback (most recent call last):
  File "linien\server\server.py", line 17, in <module>
    from autolock.autolock import Autolock
  File "C:\Users\bastian\Documents\programming\linien\linien\server\autolock\autolock.py", line 5, in <module>
    from linien.common import (
ModuleNotFoundError: No module named 'linien'
```

I'm using conda on Windows 10. Do working directories behave differently here? Do you have linien installed in your example? In this case, it wouldn't load from the parent directory but from the installed package. Maybe check `linien.__version__`.

hermitdemschoenenleben, 2021-09-03:
ah, maybe I found the problem: `server.py` does this:

```
sys.path += ["../../"]
```

which allows to find the `linien` module on linux but not on windows. Does replacing it with

```
sys.path += [os.path.join('..', '..')]
```

work?

bleykauf, 2021-09-03:
Great, that was indeed the problem! 

Well, your solution kind of works, i.e. I can now start the fake server from within the server folder but not from other folders. 

This is due to fact that when you run a python file imports relative to the file's location don't really work (see https://stackoverflow.com/a/14132912/2750945). 

I now remember that I have dealt with a similar problem in the past and I know that this can be resolved. I will have a look at this next week and propose a fix.

bleykauf, 2021-09-07:
Closed via https://github.com/hermitdemschoenenleben/linien/pull/188.

PR #188 (2021-09-07): Fix relative imports in `server.py`.
Closes #184.

~~I set `flake8` as the base branch to only show the important changes. Not sure if this can later be changed to `master` once flake8 ius merged, let's see...~~ Base branch can be changed later to `master`.
closed 2021-09-07 by bleykauf