linien-org/linien #198: A lot of objects are called connections

bleykauf, 2021-09-20:
I noticed that there are a lot of things called connections/connected etc., for example

```
        self.t.connected.connect(connected)
```

where `t` is a `ConnectionThread`. 

 This makes things a little confusing, espacially since pyqt also uses `connect` as a method name. 

I propose to use `establish_connection` and `connection_established` for connections between client and server, use `client` instead of `conn` and `connection` where appropriate.

bleykauf, 2021-09-20:
@hermitdemschoenenleben: any specific reason, why it is `connected`, not `connect` in 
https://github.com/hermitdemschoenenleben/linien/blob/master/linien/gui/app.py#L49-L55 ? (Should be renamed to `establish_connection`.

```
    def connected(self, client, parameters, control):
        self.device_manager.hide()
        self.main_window.show(client.host, client.device["name"])

        self.connection = client
        self.control = control
        self.parameters = parameters
```

and 

https://github.com/hermitdemschoenenleben/linien/blob/master/linien/gui/ui/device_manager.py#L79-L82

```
    def connected(self, connection, parameters, control):
        self.device_manager.hide()
        self.main_window.show(connection.host, connection.device["name"])

        self.connection = connection
        self.control = control
        self.parameters = parameters
```

I also don't see why `parameters` and `control` is its own attribute rather than just using `connection.parameters`. I think this would make things clearer to always work on the `connection` object (which should be renamed to `client`.).

bleykauf, 2021-09-20:
Also, is there a specific reason why there is a separate `RawPYRCClient`. I think it would be clearer to integrate everything into `LinienClient`. The latter basically just adds error handling to the `_connect` as far as I understand.

hermitdemschoenenleben, 2021-09-24:
> I propose to use establish_connection and connection_established for connections between client and server, use client instead of conn and connection where appropriate.

you're right that all those "connection"s are quite confusing and that `client` would be a better choice!

For https://github.com/hermitdemschoenenleben/linien/blob/master/linien/gui/app.py#L49-L55 I think `connection_established` would be a better choice rather than `establish_connection` because it doesn't initiate a connection but just updates the state after a connection has been established.

> I also don't see why parameters and control is its own attribute rather than just using connection.parameters. I think this would make things clearer to always work on the connection object (which should be renamed to client.).

The only reason for this is laziness :) Problem is that
`
self.connection.parameters.psd_acquisition_max_decimation.value
`

is quite a large number of characters just to access a parameter's value. But you're right that it makes things clearer to always work on the connection object, so go ahead :) Maybe it makes sense though to use a local variable
`
params = self.connection.parameters
`
in functions that access lots of parameters, but this is probably a matter of taste.

> Also, is there a specific reason why there is a separate RawPYRCClient. I think it would be clearer to integrate everything into
> LinienClient. The latter basically just adds error handling to the _connect as far as I understand.

I think this separation makes sense: `LinienClient` is a high-level interface that should be used if you write a simple script that tries to access Linien on a redpitaya. It includes some magic like automatically starting the server if not running and trying to reconnect several times. This magic is exactly the behavior that an experimentator in the lab wants to have.

However, if somebody wants to write a more sophisticated library, they may want more control about what is actually happening: automatic error handling, printing to console and retrying may not be the behavior the programmer wants. In this case, they should use `RawRPYCClient`. What do you think?

bleykauf, 2021-12-21:
Closed via #200.

closed 2022-01-20 by bleykauf