linien-org/linien #227: Distorted modulation signal

jutiszma, 2021-12-17:
[image]

Looking at the modulation signal (fast out one) without the sweep and control signal (fast out two) at low frequencies (500kHz) on an oscilloscope, I can observe additional amplitude modulation on the rising slope.

bleykauf, 2021-12-17:
To narrow down the issue: does this behaviour persist when going to higher
frequencies?

jutiszma, 2021-12-17:
Going to higher frequencies (5MHz) the modulation seems to extend over the whole period
[image]
.

bleykauf, 2022-01-12:
Just tested with my RedPitaya and could not reproduce the issue:

[image]

I would assume that this is rather a problem with your specific device.

closed 2022-01-12 by bleykauf