linien-org/linien #230: how to realize automatic PID parameter optimization

Chenelusive, 2021-12-27:


bleykauf, 2021-12-27:
This is currently not possible. We have thought about adding this before, see #102. However, it is not straightforward since it is not immediately clear what an optinized parameter set is: there might be different requirements regarding control bandwidth etc.

You can measure the closed loop PSD, though, by clicking the green button on the top of the right panel after locking the laser and change the PID parameters manually.

closed 2021-12-27 by bleykauf

Chenelusive, 2021-12-28:
Thanks for the reply,then what does the pid-optimization in server mean?

bleykauf, 2021-12-28:
It performs the PSD calculations etc. which in turn can be used to optimize the PID parameters manually as mentioned above. Admittedly, this is not the best name for what it does.