linien-org/linien #268: Possible failed installing the linien server

KaifengC, 2022-04-18:
I tried to run the linien for the first time on my Redpitaya STEM-125-14.

The connection to my Redpitaya was checked first since I can access to it and use it as an oscilloscope as usual.
I also tested its connection to the Internet by SSH into the Redpitaya and ping some websites, worked as well.

Then I run "linien-windows-0.5.3.exe", after setting up the IP address, user name (root as default), it tells me it needs to download and install the linien server. It looks totally good here but then after the program restart by itself (I guess the installation is supposed to be done), connecting to the device will give an error message, said:"Unable to connect to the device. If you are connecting to by hostname, please try using IP address instead". In the meantime, I can still connect to Redpitaya using my web browser and use it as an oscilloscope, proved the device itself and the connection are both fine.

My guess is that the installation of the linien server somehow failed. This could be blamed for the unstable internet connection. But shouldn't it give us a chance to retry the installation?

And could someone suggest a method of troubleshooting? Thanks.

bleykauf, 2022-04-19:
Thanks for reporting this issue.

First of all: you did use the ip adress and not the URL (`*.local`)?

In order to narrow down what the problem is, I would like you to try this:

SSH into your RedPitaya and run `screen -ls` and check if there is a screen session named `linien-server`. If so, please attach this session by running `screen -r linien-server`. If it says `ping [some number]` the server should be running fine. If not there is probably some error message that I would ask you to copy and paste here.

I also opened an issue regarding logging of these error messages for easier debugging in the future, #269.

KaifengC, 2022-04-20:
>First of all: you did use the ip adress and not the URL (*.local)?

I am using the IP address correctly

The server is not running based on the following response:

```bash

root@rp-f06707:~# screen -ls
No Sockets found in /var/run/screen/S-root.
```

To help debugging, I reflashed the SD card so that I can re-installed the server. This time I took some screen shots before the program restart. Here is what I saw:

[image]

[image]

I believe the program has some trouble cloning `pyrp3` and `midio-tools` from github. This is reasonable since I often need to use a proxy to connect to Github. To verify that the installation of some packages are failed, I listed out all the packages:

```bash

root@rp-f06707:~# pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
bkcharts (0.2)
bleach (1.4.2)
bokeh (0.12.7)
cffi (1.5.2)
click (7.1.2)
cma (3.2.2)
cycler (0.9.0)
decorator (4.0.6)
entrypoints (0.2.3)
html5lib (0.999)
i2cdev (1.2.4)
ipykernel (4.6.1)
ipython (6.1.0)
ipython-genutils (0.2.0)
ipywidgets (7.0.0)
jedi (0.10.2)
Jinja2 (2.8)
jsonschema (2.5.1)
jupyter (1.0.0)
jupyter-client (5.1.0)
jupyter-console (5.2.0)
jupyter-core (4.3.0)
jupyter-declarativewidgets (0.7.0)
libiio (0.10)
linien-server (0.5.3)
MarkupSafe (0.23)
matplotlib (1.5.1)
meson (0.42.0)
mistune (0.7.4)
myhdl (0.11)
nbconvert (5.3.0)
nbformat (4.4.0)
notebook (5.0.0)
numpy (1.11.0)
pandas (0.17.1)
pandocfilters (1.4.2)
path.py (8.1.2)
pexpect (4.0.1)
pickleshare (0.5)
pip (9.0.1)
plumbum (1.7.2)
ply (3.7)
prompt-toolkit (1.0.15)
ptyprocess (0.5)
pycparser (2.14)
pycurl (7.43.0)
pyfdt (0.3)
Pygments (2.1)
pygobject (3.20.0)
pylpsd (0.1.4)
pyparsing (2.0.3)
python-apt (1.1.0b1)
python-dateutil (2.4.2)
python-periphery (1.1.0)
pytz (2014.10)
pyudev (0.21.0)
pyvcd (0.1.3)
PyYAML (3.12)
pyzmq (15.2.0)
qtconsole (4.3.1)
redpitaya (0.98, /home/jupyter/RedPitaya)
rpyc (4.1.5)
scipy (0.17.0)
setuptools (20.7.0)
simplegeneric (0.8.1)
six (1.10.0)
smbus2 (0.2.0)
terminado (0.5)
testpath (0.3.1)
tornado (4.5.2)
traitlets (4.3.2)
wcwidth (0.1.6)
wheel (0.29.0)
widgetsnbextension (3.0.2)
You are using pip version 9.0.1, however version 22.0.4 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
root@rp-f06707:~#
```

I'm going to set up a proxy inside my root for a better connection to github. I'm also thinking is it possible to install some of the package manually without Internet connection on Redpitaya? I understand there are issues about the licenses, but providing some instructions for the user who want to install things step by step seems pretty reasonable.

bleykauf, 2022-04-20:
That does indeed look like an issue with the connection to github: do you know if this is a specific restriction for accessing github from China?

In any case, the installation process is triggered by [linien/server/linien_install_requirements.sh ](https://github.com/linien-org/linien/blob/99b57f674e14c985159f31623ee4fe8894347dc9/linien/server/linien_install_requirements.sh) where you can find the required packages. I do plan to upload at least pyrp3 to pypi and will look into a way to package mdio-tools with linien; this is however currently not a priority.

In case you cannot connect to github from your RedPitaya, I would suggest to manually copy over the two repositories via `scp` or something similar and then trigger the commands from the `linien_install_requirements.sh` manually.

Please let me know if you were able to resolve the issue, I can then add this to the Troubeshooting section of the readme.

closed 2022-05-05 by bleykauf