linien-org/linien #272: linien-server issue: libmonitor.so not found?

lelezeik, 2022-05-02:
Hi Linien peeps!

So I've been having this issue when I try to connect to my redpitaya. Running linien and screening to linien-server, it seems to be crashing when it has to execute `libmonitor.so`. Here's what it looks like: 

[image]

Anyone faced a similar issue? 

Thanks!

lelezeik, 2022-05-09:
Thanks for the reply @bleykauf 
I did something similar but from this repository ` https://github.com/clade/PyRedPitaya.git`  as it also contains the libmonitor.so and that fixed the problem. 
 Linien-Client installed the server automatically.

bleykauf, 2022-05-09:
Thanks for the reply. pyrp3 is in fact a fork of the package you installed. Apparently some recent changes on our side caused the issue. I am currently looking into it. In the future I am planning to make pyrp3 installable via pip, including the `libmonitor.so`.

Here is the current output of `make`:

```
root@rp-f09587:/tmp/pyrp3/monitor# make
cp shared//include/redpitaya/version.h .
cp: cannot stat 'shared//include/redpitaya/version.h': No such file or directory
Makefile:84: recipe for target 'version.h' failed
make: *** [version.h] Error 1
```

bleykauf, 2022-05-09:
So this should be resolved via https://github.com/linien-org/pyrp3/pull/1.

The version history of https://github.com/clade/PyRedPitaya and https://github.com/hermitdemschoenenleben/pyrp3 (base of https://github.com/linien-org/pyrp3) is actually a little unclear to @hermitdemschoenenleben and me: they all share common code but the exact origin of the inital commit for pyrp3 is not entirely clear. Probably best to base on https://github.com/clade/PyRedPitaya or use that package directly.

@lelezeik: did you simply run the same commands with the other repository or install it via `pip3`?

bleykauf, 2022-05-09:
Also got confirmation from a user via email that this resolved the issue.

closed 2022-05-09 by bleykauf