linien-org/linien #275: New hardware platform

kaolpr, 2022-05-26:
At WUT we're developing Sinara Open Hardware ecosystem and recently received prototypes of the board that may be of interest to you. The project is called Fast Servo and is designed in a 3U form factor to be used in crate or standalone. It features 2ch 125 MSPS 16b ADC and 2ch 125 MSPS 14b DAC and is based on Trenz Xilinx Zynq module. After we're done with testing, it will be available for purchase from our industrial partners. 

For more information see project [Wiki](https://github.com/sinara-hw/Fast_Servo/wiki). 

If you are interested in using Fast Servo with linien we can contribute and port it to our board.

bleykauf, 2022-06-01:
We are definitely interested in a port of Linien to Sinara hardware! I have personally not used Sinara/Artiq myself but some of my colleagues do and I might do so myself in future projects.

I have planned for some time to separate the different components of Linien more clearly, i.e. having separate packages for server, client and GUI and clearly separating migen/verilog/FPGA related code. I have so far not gotten around of actually doing this (other than some research into python namespace packages) but I think doing this would make porting Linien to other platforms much easier.

We should also consider which parts of the code are actually hardware dependent and separate those in order to avoid diverging codebases.

kaolpr, 2022-06-01:
I believe we could help you with this refactorization while porting to Fast Servo. We will get into the project details and maybe then we could arrange some meeting to go into details and setup a work plan?

bleykauf, 2022-06-01:
That sounds great!

I think I can set aside some time tomorrow to get this started. Also happy to have a meeting once you got an idea about the code.

bleykauf, 2022-06-02:
Started today with the refactor: https://github.com/linien-org/linien/pull/277. This is not yet finished and probably won't work yet, will continue as time permits.

bleykauf, 2022-06-22:
The PR is still WIP but I think I am on a good way and the components of linien are now more clearly separated (did not use a namespace package after all).

There will be 5 packages, linien-common, linien-server, linien-client and linien-gui that will be put on pypi. There is also the gateware which is not a package but is transpiled using migen. All packages and the gateware rely on linien-common. 

The easiest way is probably to install packages via pip in editable mode, starting with `pip install -e .` from the `linien-common` directory and then install the other packages the same way as needed.

As for dependencies: All packages and the gateware rely on `linien-common`. `linien-gui` depends on `linien-client`. `linien-server` has some additional dependencies installed via `linien_install_requirements.sh`. This includes `pyrp3` and `mdio-tool`. Specifically, `pyrp3` is used in `acquisition_process.py` and `mdio-tool` is used in the `linien_start_server.sh` to turn of ethernet blinking on the RedPitaya. `csrmap.py` is generated via the gateware. 

What I have tested succesfully:
* building the FPGA image via `build_fpga_image.sh`
* building standalone binaries of `linien-gui` with `pyinstaller pyinstaller.spec` on both Linux and Windows
* running `linien-server` on the RedPitaya and connecting to it
* running `pytest .`: the `linien-` packages need to be installed via pip
* uploading to testpypi

Things that do not work yet:
* github Actions
* installation of the development version via the client/gui
   * currently working on this
   * want to use `.devX` versions according to [PEP440](https://peps.python.org/pep-0440/) for development versions
   * want to install via `pip install -e .` so everything behaves the same, independent of whether dev or stable version is used

This needs to be updated but most things should still be valid: https://github.com/linien-org/linien/wiki/Development
Most notably, each package now has its own `_version.py` that has to be changed individually. I did not find a way to unify this.

@kaolpr : Feel free to check out #277 :) The most relevant things for a port should already be clear.

bleykauf, 2022-06-22:
Also note, that on the RP, we are unfortunately limited to use Python 3.5, see https://github.com/RedPitaya/RedPitaya/issues/233

kaolpr, 2022-06-28:
@bleykauf Thank you for quick and extensive action!

We're currently debugging the prototype of FastServo. As soon as we're done with it we'll dive into linien.

bleykauf, 2023-03-21:
@kaolpr just checking in, is this still something you consider? Separation between the different components is now much clearer.

kaolpr, 2023-03-25:
We do have a working prototype. I'll look into it and see when we can submit PR.

linuswck, 2024-12-03:
@bleykauf 

I am now porting the linien gateware and its software elements to the [Fast Servo Platform](https://git.m-labs.hk/M-Labs/nix-servo). We are porting release 1.0.2. I have discovered some problems related to the gateware and/or software, and I want some clarifications on the expected behaviors of linien.

1. When I compiled the linien gateware plus our DAC and ADC interface gateware, I saw a lot of timing violations in the linien DSP signal chains, and it does not meet the timing. They are mainly caused by the lack of output pipeline register for the inferred DSP48E1 cells. Do these timing violations also exist on the unmodified version of Linien?
2. To make the design meet timing, I have added some pipeline registers at the output of the DSP48E1 and enabled global retiming during synthesis. Any suggestions on how to verify its full functionalities on hardware? Any suggested setup?
3. I have set up a laser spectroscopy lock setup for verifying its PID and its autolock feature. FAST OUT 1 (Sweep and Modulation Signal) -> Temperature Stabilized LD -> Reference Gas Cell -> PD -> FAST IN 1. I can see the absorption spectrum.
[image]
However, when I try the auto lock feature(using simple mode), apparently, the control signal and its control history have the modulation signal mixed in. Changing the modulation amplitude also directly affects the control signal readouts.
[image]
(I changed the modulation amplitude: 2.0V -> 1.0V -> 0.0V -> 2.0V in this screenshot.)

I don't think this is the expected behavior. Which gateware control signal does the linien-gui display when pid is locked?

Thank you for your help in advance.