linien-org/linien #282: Two sets of PID for feedback on both OUT 1 and OUT 2.

MegaKell, 2022-08-03:
Slow control via analog out is too noisy. I think it would be cool to either send the slow to out 1 or have a separate set of PID controls for OUT 1 and OUT 2 for systems with two stages of feedback (like PZT and AOM).

bleykauf, 2022-08-11:
I think that is a good idea! Currently, I do not have a whole lot of capacity to work on additional features. You might want to have a look at `linien.py` on the server side. I guess that it should be relatively straight-forward to change the output of the slow integrator.

MegaKell, 2022-08-11:
That is a good idea, thanks! I have mostly been using the GUI you developed and really like it! Thanks for sharing.

PR #284 (2023-02-28): Ability to output fast and slow PID on fast-DACs
Modified bitstream and GUI to allow outputting slow-PID to fast OUT1/OUT2 channels.

Useful for ECDL locking or other systems with two feedback channels, this should resolve the issue described in:
resolve #282 

Notes:
* remaining bug: when disabling slow_PID, the GUI element flips to "ANALOG_OUT0" (slow_PID output is off, though)
* merging back into master (instead of develop) because of build errors when testing develop branch.