linien-org/linien #283: connection of device

atomarray49, 2022-08-11:
Hi
   I am trying to connect the device with linien-windows-0.5.3. But I meet the problem unable to connect to the device neither by host name or IP address. I do not know the reason.
  I need the help.

bleykauf, 2022-08-11:
First check whether you can ping it by
executing

```bash
ping rp-f0xxxx.local
```

in a command line. If this works, check whether you can connect via SSH:

```bash
ssh rp-f0xxxx.local
```

on the command line. If this is successful, in order to to  check whether the `linien-server` is running, first confirm that there is a running `screen` session with the name `linien-server` by  executing `screen -ls`. If that is the case attach it by running `screen -r linien-server`. If any errors occurred on the server side, they will be displayed here. Please provide the output in your bug report.

bleykauf, 2022-10-09:
Closing since inactive.

closed 2022-10-09 by bleykauf