linien-org/linien #290: linien.client python module not found when running the gui

doronbehar, 2022-10-08:
Hello!

I'm packaging this application for NixOS, and I get the following error when running the `linien` executable after installing the gui via `setup_gui.py`:

```
Traceback (most recent call last):
  File "/nix/store/l6bb916d6pww2nbcxiyl84arsi5iqpgb-python3.10-linien-gui-0.5.3/bin/.linien-wrapped", line 6, in <module>
    from linien.gui.app import run_application
  File "/nix/store/l6bb916d6pww2nbcxiyl84arsi5iqpgb-python3.10-linien-gui-0.5.3/lib/python3.10/site-packages/linien/gui/app.py", line 36, in <module>
    from linien.gui.ui.device_manager import DeviceManager
  File "/nix/store/l6bb916d6pww2nbcxiyl84arsi5iqpgb-python3.10-linien-gui-0.5.3/lib/python3.10/site-packages/linien/gui/ui/device_manager.py", line 26, in <module>
    from linien.client.connection import LinienClient
ModuleNotFoundError: No module named 'linien.client'
```

But if I make the following small change:

```diff
diff --git i/setup_gui.py w/setup_gui.py
index b47f0fb..2b95efb 100644
--- i/setup_gui.py
+++ w/setup_gui.py
@@ -38,7 +38,7 @@ setuptools.setup(
     url="https://github.com/linien-org/linien",
     # IMPORTANT: any changes have to be made in setup_client_and_gui.py
     # of flathub repo as well
-    packages=["linien", "linien.gui", "linien.gui.ui"],
+    packages=["linien", "linien.client", "linien.gui", "linien.gui.ui"],
     classifiers=[
         "Programming Language :: Python :: 3",
         "License :: OSI Approved :: GNU General Public License v3 (GPLv3)",
```

I get the GUI launching fine. I also noticed that the `linien` pypi package tarball doesn't contain the `linien/client` directory and it's `setup_gui.py` is the same as in the git repository.

Do you have any idea why is this so?

bleykauf, 2022-10-09:
Thank you for pointing that out!

I am still in the process of restructuring the package and make the dependencies clearer, see #https://github.com/linien-org/linien/pull/277. Its been a while since I last worked on this. I hope that I will find a little bit more time to finish this during the winter months... This should then be resolved.

bleykauf, 2023-03-19:
Fixed via #277

closed 2023-03-19 by bleykauf