linien-org/linien #299: Possible memory leak with linien-server?

systemofapwne, 2022-10-26:
Recently, we observed that our Redpitaya running linien was not reachable anymore the next day via linien or direct ssh while pings still worked.
Sometimes it got resolved by "turning the redpitaya off and on again" but this mostly lead to a corrupted root-filesystem in the end so the machine refused to boot properly. Fixing the SD-card with an SD-Card reader via "fsck" on a second PC helped when having a corrupted filesystem. The SD-card itself is fine though (already replaced by 3 different cards of different brand with a fresh install of STEMLabs)

Today, while debugging, I had the idea to permanently connect to the serial console in case that anything interesting will be there once the machine "freezes" again. I will report back here, when I see something (so far nothing special yet).
However I just recently noticed after checking in via (yet working) ssh, that the machine was quite low on memory and about 85% was used by the linien server alone. Stopping and restarting the linien-server just made it consume about 10% RAM.
This now draws me towards the hypothesis, that there might be a memory leak.
The strange thing is, we did not see this before on similar setups.
The only thing that changed was, that the linien-client connecting to the redpitaya in question is running on a machine, that goes to standby-mode after 2h of inactivity and linien-client was still connected to the server when this happens (yet, no locking was in place though). The server then just kept on pinging (seen via ``screen -r linien-server``). Maybe it also caches some data "to be sent" to the client and at some point runs out of memory?
I will further try to debug this issue, once it is conclusive. Right now, it almost always takes about a full day ore more until this issue appears, so it might be quiet lengthy to debug.

systemofapwne, 2022-10-27:
I am not sure, if the memory leakage is yet caused by the client not responding anymore (probably can test in more detail tomorrow). 
But I am sure, the linien-server does indeed leak memory: https://youtu.be/_dsGmGbOsTA
There, you see, how the python process increases 0.1% of memory usage after ~ 50-60 seconds. This trend continues.
I then started a linien client on another host: https://youtu.be/RauOQpM_fXk
Here, a new fork of the linien-server is produced. When quitting the linien-client on that host, the forked process disappears.
I guess, that should also happen, when a client unexpectedly disconnects from the linien-server. Like a kill-timeout of the linien-server, just to avoid these kind of issues.
Note that this would only be a workaround, if the linien-server also would leak memory when running normally with a working client permanently connected. But if the mem-leakage only happens when a client "disappears" without telling the server "goodbye", killing the server after too many unanswered "pings" might be the solution though.

systemofapwne, 2022-12-15:
Sorry for the late update.

So linien-server definitely leaks memory, when a client stops responding without disconnecting from the linine-server first.

Our current workaround is to kill all linien instances before the machine goes to sleep. Sinnce we run Linux, this can be done pretty easily with a systemd-script like this here:

/var/systemd/system-sleep/kill_linien.sh

```
#!/bin/bash

killall -9 linien
```

Note, that this runs on any suspend event (even wakeup). But thats fine for the purpose.

Note 2: This bug definitely has to be fixed in linien-server and not cured by a workaround.

bleykauf, 2022-12-15:
Sorry for the very late reply. I recently did not have too much time to work on Linien...

Thank you very much for bringing this to our attention and providing extensive documentation and the workaround for this issue. I fully agree that this should be fixed within linien-server. I hope once the refactored version #277 is ready to be released we can have a closer look at this issue and responsibilities of the individual components of the software will be clearer.

systemofapwne, 2023-07-06:
Thank you for your reply. I will see, if we can reproduce the issue in our lab in the near future.

bleykauf, 2023-11-21:
@systemofapwne: Since I haven't heard back, I assume the error did not occur again?

systemofapwne, 2023-11-21:
Sorry, I forgot about testing it, since we had the workaround in place for some time now.
I will schedule a timer for me to check the next days! Thank you for reminding me!

systemofapwne, 2023-11-23:
I now had linien 0.8.0 running on the redpitaya for some hours, while the PC with the GUI was in standby.
With that version, I was not able to reproduce the issue anymore. The memory leak seems to be fixed.

closed 2023-11-23 by systemofapwne

bleykauf, 2023-11-23:
@systemofapwne: Thank you, glad the issue is resolved :)