linien-org/linien #300: Installation procedure of Linien development version unclear

jmatyas, 2022-12-08:
Hi, I'm trying to set up and launch Linien and I have to say that neither documentation  nor wiki development page are of much help. It is not stated clearly which installation step should be executed on a host machine and which (if any) on Red Pitaya. 

## Problem description 
I have successfully finished the following steps:

- cloning the repository (from the development wiki page); here it is stated that: 

>This ensures that local changes of the server's code are automatically uploaded to RedPitaya when you launch the client. Please note that this only works if linien-server is stopped and uninstalled from the RedPitaya
- set up the development environment (on the host PC - my assumption); I chose the pip and run: ```pip install -r requirements_dev.txt```
- running the code
  * here came the first confusion - docs tell me that there should be no running server on RedPitaya, even though at the very beginning it is stated that it should be uninstalled. No files provided by Linien repo are available on RedPitaya yet, so there is no ```linien_stop_server.sh``` script. And even if it had already been copied to RedPitaya, there is no ```screen``` command available.
  * build gateware with the use of the ```build_fpga_image.sh``` (according to Vivado build was successful); new ```linien.bin``` file appeared in the server folder

According to the development wiki now it should be the time to launch the client. Running linien/gui/app.py script launches a client successfully, I am able to add a device, however, program is unable to start a server on RedPitaya. There is a message "Error starting the server" and after a while: "ConnectionRefusedError: [Errno 111] Connection refused
Error: connection to the server could not be established".  
Looking into the code and error messages revealed that on RedPitaya there was no screen program and that there is no module named "plumbum" (see the snippet below)

```(linien) ➜  linien git:(master) ✗ python linien/gui/app.py
Linien spectroscopy lock version dev
try to connect to 192.168.0.171:18862
server is not running. Launching it!
uploading dev source code...
Error starting the server
b'Traceback (most recent call last):\n  File "<string>", line 1, in <module>\n  File "/linien/__init__.py", line 5, in <module>\n    from plumbum import colors\nImportError: No module named \'plumbum\'\n'
try to connect to 192.168.0.171:18862
```
## What has been already done

- tried connecting to the RedPitaya with IP and rp-xxxxxx.local and in both cases the behaviour was the same
- installed ```screen```
-   confirmed that there is no plumbum module on RedPitaya via SSH: 
```
root@rp-f0a162:~# python3 -c "import plumbum"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named 'plumbum' 
```
- confirmed that there is no ```linien``` module on RedPitaya

All of the above and the comment inside ```linien_start_server.sh``` pointed to me that maybe ```linien``` should be installed globally with pip.

## Expected behaviour
I would expect that following the wiki development page step by step would allow me to install Linien on Red Pitaya without such problems - either make installation script run by ```linien/gui/app.py``` foolproof or specify which steps should be run  on host PC and which on RedPitaya

jmatyas, 2022-12-12:
So, after looking into the code and a couple of tries, here are some notes that I made during the installation of development version process. This is the process that works on a new, clean RedPitaya:

- install linien-server on RedPitaya with `pip3 install linien-server` (it installs among others python dependecies needed by scripts)
- install all required programs with `linien_install_requirements.sh` 

The above are more or less the same what is performed by Linen when it discovers that "there is no Linien Server on RedPitaya", however it does not come to that conclusion on its own just by running the code with `python linien/gui/app.py`.

During the next connection to RedPitaya with Linien the following window emerges: 
[image]. 

So doing as the Linien says (on RedPitaya SSH): `cd /linien/server && chmod +x linien_stop_server.sh; pip3 uninstall linien-server` allows to finally run development version from a host computer with `python linien/gui/app.py`

bleykauf, 2022-12-12:
The whole deployment process will hopefully soon be majorly overhauled since it was rather hacky before. There are only a couple of changes I need to make to #277. I hpe I can find the time to work on this later this year. The wiki definitely seems wrong about the installation procedure, both for the current `master` branch and the `develop` and `feature` branches.

Thank you for coming back with a solution (I think it is working now, right?). Do I see correctly, that the solution was to run the `linien_install_requirements.sh`? I will indicate in the wiki that the section is incorrect and point to this issue.

jmatyas, 2022-12-12:
> Thank you for coming back with a solution (I think it is working now, right?). 

I hope it is working correctly ;). But I assume so, because the server on RedPItaya starts and client window changes and shows some "signal".

> Do I see correctly, that the solution was to run the `linien_install_requirements.sh`? 

It was running linien-server installation with `pip3 install linien-server` on RedPitaya (Python packages), then `linien_install_requirements.sh` (system level programs etc.) and finally doing what the Linien app launched on a host PC told me to do - `linien_stop_server.sh` and `pip3 uninstall linien-server`.

aimpertro, 2022-12-17:
We recently tried to set up the dev version as well and faced similar problems. Additional steps required were:

- In the linien_start_server.sh file, remove the $ signs in lines 19 and 21 preceding the strings after the 'stuff' argument.
- The gateware (linien.bin) that is currently in the main branch of the repo appears to be corrupted and causes the RedPitaya to crash and reboot. A solution is to install the linien-server pip package and extract the gateware from there, replacing the gateware in the cloned repo.

After doing this (and installing the requirements mentioned above) we got it working.

jmatyas, 2022-12-17:
>     * The gateware (linien.bin) that is currently in the main branch of the repo appears to be corrupted and causes the RedPitaya to crash and reboot. A solution is to install the linien-server pip package and extract the gateware from there, replacing the gateware in the cloned repo.

I didn't encounter that problem, but I built gateware on my own. However I used the source code from the repo - didn't test it's functional behaviour, but it certainly didn't cause RedPitaya crashing or rebooting

bleykauf, 2022-12-17:
@aimpertro: Did you make sure to have git lfs set up? Otherwise the linien.bin will only contain some hash value.

aimpertro, 2022-12-17:
@bleykauf I believe so, yes. The binary was around 2MB in size. But cannot entirely reproduce this anymore, we tried out too many things in the meantime. 

One more thing: When running the server remotely (starting only the acquisition_process.py on the RP), the gateware does not seem to be flashed onto the FPGA but needs to be manually loaded by copying the binary to /dev/xdevcfg. If this is not done, the server and the gui start happily but the scope is not triggered. This would also be good to add to the dev wiki.

bleykauf, 2023-02-27:
Updated the wiki for the soon-to-be-released version 0.6.0. I hope the instructions are now clearer.

I personally could not get the server to run remotely, see #309. I personallly never used this specific functionality before...

closed 2023-02-27 by bleykauf