linien-org/linien #326: bugs in oscilloscope, display and PSD

wwcphy, 2023-03-29:
There are several small bugs I observed:

### Oscillosopce x axis

1. x axis label for control history (maybe not a bug)
When locked, the x axis label doesn't match the time scale of the control history (green trace). Maybe that is the time scale for the control (red trace)? The history below is a ~10min lock.
[image]

2. Fast mode error signal not showing
I realize someone already reported this in #314. We got stuck on this for a while. But the solution in #314 works. Maybe it is better to mention this in README?

### PSD measurement
3. PSD curve not physical
In many measurements, I get the following spectrum. There is a sudden jump at 10^2 Hz, sometimes also at ~2.5*10^4 Hz. They seem not very reasonable to us. I checked the lock status during PSD measurement, it was not railing at any limits and behaved normally. 
[image]

Thank you for your project!

wwcphy, 2023-04-04:
"voltages should be displayed rather than the time (which typtically has no significance to the user)"

Actually, we want control signal/error signal versus time more than voltage. Since there is not extra port from RP for us to monitor signals on an external oscilloscope, it is important we see them versus time on the built-in oscilloscope. 
I don't understand why you want to put voltage in x axis?

bleykauf, 2023-04-05:
Sorry, I misread. I would like to put the x-axis as a voltage when the sweep is active since you can now also control the sweep amplitude. For the locked state there should be some indicator for the time (history), of course.

On a related note: In your screenshot, I do not see the sweep control slider above the plot. Is this a manual change you made?

GeorgeSchwartz, 2023-06-23:
I would like to follow up on this thread. I would like to see the Control/Error Voltage as a function of time. Is there a way to customize the x-axis? For example, so wwcphy's first plot shows 10 minutes as opposed to .0001152?

Alternatively, when exporting the data, is there a default time interval between consecutive readings? The .json file give all the voltages for the 'control signal', 'error signal', etc. but doesn't specify how often these voltages are read. Is this something the user can control?

Thanks for the program!