linien-org/linien #330: Run linien-server as a systemd service?

bleykauf, 2023-04-04:
As suggested by @doronbehar in https://github.com/linien-org/linien/pull/277#issuecomment-1272506334 , it would look like this 

```
# linien.service
[Unit]
Description=Spectroscopy lock server for RedPitaya
Wants=network-online.target
After=network-online.target

[Service]
# Change this value either with `systemctl edit linien-server.service`, run
# `man systemctl(1)` to read documentation for this command. Perhaps this
# should be read from a file such as `/opt/linien/port` instead.
Environment="LINIEN_PORT=8000"
ExecStart=/usr/bin/env python3 /usr/lib/python3.10/site-packages/linien/server/server.py
# stop / start ethernet blinking using: mdio-tool[1], see:
# https://github.com/RedPitaya/RedPitaya/issues/205
#
# [1]: https://github.com/linien-org/mdio-tool
ExecStartPre=/usr/bin/mdio-tool w eth0 0x1b 0x0000
ExecStopPost=/usr/bin/mdio-tool w eth0 0x1b 0x0f00

[Install]
WantedBy=multi-user.target

# vim:ft=systemd
```

doronbehar, 2023-04-04:
Duplicate of #293 :).

bleykauf, 2023-04-04:
A thanks, I thought I forgot about that!

closed 2023-04-04 by bleykauf