linien-org/linien #334: Installation of linien-server fails on RedPitaya 2.00 os release

blupgnup, 2023-05-16:
Installation of Linien server through the windows client is not successful.
This bug appears when using Linien 7.0 and RedPitaya OS 2.00 release (Beta)

This seems to be due to missing package : libutempter0 and screen.
Installer is stuck while running linien_install requirements.sh as package manager is waiting for confirmation for the installation of those new packages (which we cannot give through remove installer...)

Installing both packages before running the installer does the trick.
Forcing the acceptation of the package installation within the script should also work...

You might then need to remove linien-server (pip uninstall linien-server).
Server seems to start without that but I did not test if some packages might be missing if the linien_install_requirements script is not run to the end...

bleykauf, 2023-05-16:
Thanks for the report. Indeed, the installation commands should add the appropriate action to not require user interaction.

To clarify: did not manage to get it to work in the end? So far, I have been unable to get data back from the RP, see #322.

blupgnup, 2023-05-17:
Sorry, it was not clear, indeed.
Installation appears to be successful but when starting Linien, no data seems to be retrieved from the RP.
I had to switch back to the latest stable release for now as I do not had the time to check what was wrong.

Just reporting because I think the script will have to be adapted anyway...

bleykauf, 2023-05-17:
Thanks for the update.

Added `-y` option in the `linien_install_requirements.sh` in #335 that should resolve the installation issue. Unfortunately, for now Linien will only work on RedPitaya OS 1.04 (added note about this in the README).

bleykauf, 2023-07-06:
The new release [0.8.0](https://github.com/linien-org/linien/releases/tag/v0.8.0) is compatible with RedPitaya OS 2.0. Closed.

PR #335 (2023-05-24): Improve installation script and add note about RedPitaya OS 2.00
Fixes #334
closed 2023-07-06 by bleykauf