linien-org/linien #337: The GUI doesn't show the spectra at all

SoraUmika, 2023-06-14:
After successfully installing Linien and having the atomic spectra signal loaded to input 1 I wasn't able to see any signal at all. 
 I am scanning the atomic transition externally and using Red Pitaya's preinstall oscilloscope I was able to see the signal without a problem, but when connected to the Lilnien GUI, nothing shows up on the viewing screen

SoraUmika, 2023-06-14:
Not sure if this is a possibility but the software server seems to be checking updates every time it is turned on. However, in our setup, our red pitaya does not have an internet connection. I recalled receiving an error once related to a network issue.

bleykauf, 2023-06-15:
If you do not have internet connection on the RedPitaya, you will not be able to install linien-server on it. To check if the server is running, do the following:

Connect via SSH,

```bash
ssh rp-f0xxxx.local
```

on the command line.Then, to check whether the `linien-server` is running, first confirm that there is a running `screen` session with the name `linien-server` by  executing `screen -ls`. If that is the case attach it by running `screen -r linien-server`. If any errors occurred on the server side, they will  be displayed here.

You can also manually run the server with `linien-server`.

SoraUmika, 2023-06-16:
The server is correctly installed and I am running on Red pitaya 2.0 with linien verz =0.7
I received the following warnings upon turning on the server

/usr/lib/python3/dist-packages/scipy/signal/_filter_design.py:1709: BadCoefficients: Badly conditioned filter coefficients (numerator): the results may be meaningless
  warnings.warn("Badly conditioned filter coefficients (numerator): the "
/usr/lib/python3/dist-packages/scipy/signal/_filter_design.py:1083: RuntimeWarning: invalid value encountered in true_divide
  b /= b[0]

76839ao, 2023-06-17:
Dear SoraUmika,
Recently, I also encountered a problem similar to yours. When I connected the error signal to input1 of redpitaya, I can't see the signal on the linien GUI. But when I connected the error signal to input2, I can see the signal on the linien GUI. And, when I connected the error signal to input1 of redpitaya using redpitaya's preinstall oscilloscope, I can see the signal. So, this shows that the input1 channel of redpitaya is not broken.
Have you solved this problem now? 
Looking forward to communicating with you.

SoraUmika, 2023-06-17:
I haven't fixed it yet, i suspect it might be an issue with the OS, so I will try to downgrade the OS. There is a known working version of red pitaya OS vers 1.04 and Linien 0.5.3 according to another person I know that is using this system. In my case, I can't see either channel 1 or channel 2 signals. I also monitor the output, which is the sweep and modulation output, the sweep does output something but definitely not a sweep and the signal is low, it could be some kind of noise. The modulation signal is just zero despite me tweaking the parameters in the GUI.

Another thing I might also consider trying is to look for the function that reads channel 1 and channel 2 output in linien and see if I can deduce the problem there, but for now downgrading the OS might look the most promising

SoraUmika, 2023-06-18:
I fixed it, I should've known this earlier but I believe Linien is running C code underneath, in that case you'll need to download the OS that supports C language system. Its on the official website and choose the one suitable for your board.

closed 2023-06-18 by SoraUmika

76839ao, 2023-06-18:
Hello,
I have solved the problem.  Please make sure that the modulation frequency/amplitude are set to 0.

Thanks.

Ao

Closed #337 as completed.