linien-org/linien #364: Linien `1.0.0` server instance not starting after connecting

ograsdijk, 2023-12-07:
Red Pitaya OS `2.00-18-stable`
Linien version `1.0.0`

When starting Linien GUI a screen session is created on the red pitaya, but the server isn't actually started. After several connection attempts Linien GUI throws a connection refused exception.

Attaching the screen session shows that nothing is actually executed. After manually starting `linien-server` inside the screen session the GUI behaves as expected.

The issue persists after reflashing the SD card and reinstalling Linien on the Red Pitaya. Manually installing `linien-server` and `screen` on a flashed SD card results in the same issue.

bleykauf, 2023-12-07:
Thanks for your bug report.

This issue does not occur for me (same OS and linien version). Are you using the Windows executable for installation of the remote or are you manually installing it via pip? If you installed via pip, you probably have to run `linien_install_requirements.sh`.

Could you please run `linien_start_server.sh`on the Red Pitaya and report back if the screen session is started and is running linien-server?

ograsdijk, 2023-12-13:
I manually installed linien via pip on Windows. I did run `linien_install_requirements.sh` on the red pitaya.

When running `linien_start_server.sh` the screen session was started, but nothing happened in the screen session.

Issue was fixed by re-imaging the red pitaya SD card and using the Windows executable for the initial installation. Afterward the pip installed linien version works just fine.

bleykauf, 2023-12-13:
Thank you for reporting back! I will reopen since I did specifically not test the installation process with the pip-installed version. I do not see right now what the problem could be but I will have a look.

bleykauf, 2023-12-22:
Thanks again @ograsdijk. I could reproduce the issue on my end. While I do not fully understand the cause of the problem, I simplified the startup script and things work with the latest release v1.0.1 (#370)

PR #369 (2023-12-21): Fix linien-server startup
Simplify the startup script and fix #364.
PR #370 (2023-12-22): Release v1.0.1
Fixes #364.
closed 2023-12-22 by bleykauf