linien-org/linien #394: Cannot connect to Redpitaya 250-12 with OS 2.00-30

Zacstein, 2024-05-04:
1. The executable file linien-2.0.0.exe can successfully install `linien-server` on the redpitaya.
2. `ping` and `ssh` success, but the connection always show time-out error.
3. The command `screen -ls` shows no sockets, and after the command `linien-server start`, `screen -ls` still shows no sockets. Additionally, `screen -r linien-server` shows "not matching" error.

I have no idea about how to fix this problem. Thank you in advance!

chengshufn, 2024-05-10:
+1, I'm using Redpitaya STEMlab 125-14 LN.

bleykauf, 2024-05-13:
I forgot to update the README. Linien 2.0 uses systemd rather than screen for running the server. Please run `linien-server status` or check `/root/.local/share/linien/linien.log` for debugging.

You can also try to run the server directly with `linein-server run`.

Zacstein, 2024-05-14:
> I forgot to update the README. Linien 2.0 uses systemd rather than screen for running the server. Please run `linien-server status` or check `/root/.local/share/linien/linien.log` for debugging.
> 
> You can also try to run the server directly with `linein-server run`.

[image]

Zacstein, 2024-05-14:
And I check the file linien.log, which is always empty.

bleykauf, 2024-05-14:
OK, so there is something wrong when deploying the gateware... 

Could you please verify that the SH256 checksum of the gateware is `47bba375ba97bb1e9de0261844b2e527521bea795e3f6233e65247a0ba4a41bf`?

You can do this by running

`sha256sum /usr/local/lib/python3.10/dist-packages/linien_server/gateware.bin`

Also, I just uploaded a new version yesterday. The old one had a severe bug related to the autolock.

Zacstein, 2024-05-15:
I change to use the new version linien-2.0.2.exe, but encounter new bug.
[image]

And I also verify the SH256 checkesum of the gateware, which is correct.
[image]

bleykauf, 2024-05-15:
Looks like the `parameters.json` got corrupted due to the crash when uploading the gateware. Please delete it, `rm /root/.local/share/linien/parameters.json`, and try again.

Zacstein, 2024-05-15:
Thanks for your quick reply! I delete the json file, but the old bug apears.....
[image]

[image]

bleykauf, 2024-05-15:
That is odd. I did test specifically test with the same OS version and STEMlab 125-14 LN without any issue. I would not rule out some hardware issue. Do you happen to have another RP lying around that you could test on?