linien-org/linien #405: Can't install the server on redpitaya

felixhoff, 2024-05-28:
Thank you for reporting a bug!

I'm working with Linien 2.0.2
In the case that you are  having trouble with connecting to the linien-server running on the RedPitaya, please try the following first and  include any errors and the console output in your bug report:

I can ping and connect via ssh.

I am trying to install the server with the following python code:
`from linien_client.device import Device
from linien_client.deploy import install_remote_server

device = Device(
    host="rp-f0c345.local"
)
install_remote_server(device)`

And I get this output:
`C:\Users\PC\anaconda3\Lib\site-packages\paramiko\transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated and will be removed in a future release
  "class": algorithms.Blowfish,
>> linien-server stop
bash: line 1: linien-server: command not found
>> pip3 uninstall linien-server -y
WARNING: Skipping linien-server as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
linien_client.deploy           DEBUG    Sucesfully executed 'pip3 uninstall linien-server -y'
>> pip3 uninstall linien-common -y
WARNING: Skipping linien-common as it is not installed.
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
linien_client.deploy           DEBUG    Sucesfully executed 'pip3 uninstall linien-common -y'
>> pip3 install linien-server==2.0.2 --no-cache-dir
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f55f8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f56e8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f5988>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f5a60>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f5b38>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
ERROR: Could not find a version that satisfies the requirement linien-server==2.0.2 (from versions: none)
ERROR: No matching distribution found for linien-server==2.0.2`

I tried to install it with PuTTY, but I get this output : 
`pip3 install linien-server
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f2490>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f2580>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f2760>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f2838>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xb53f2910>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/linien-server/
ERROR: Could not find a version that satisfies the requirement linien-server (from versions: none)
ERROR: No matching distribution found for linien-server
`

By the way, it seems the GUI isn't working for me. I just ran pip install linien-gui, and then try to run the command linien and I get this output :
`C:\Users\PC>linien
C:\Users\PC\anaconda3\Lib\site-packages\paramiko\transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated and will be removed in a future release
  "class": algorithms.Blowfish,
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\PC\anaconda3\Scripts\linien.exe\__main__.py", line 7, in <module>
  File "C:\Users\PC\anaconda3\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\linien_gui\app.py", line 116, in main
    app = LinienApp(sys.argv)
          ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\linien_gui\app.py", line 48, in __init__
    self.main_window = MainWindow()
                       ^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\linien_gui\ui\main_window.py", line 87, in __init__
    uic.loadUi(UI_PATH / "main_window.ui", self)
  File "C:\Users\PC\anaconda3\Lib\site-packages\PyQt5\uic\__init__.py", line 241, in loadUi
    return DynamicUILoader(package).loadUi(uifile, baseinstance, resource_suffix)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\PyQt5\uic\Loader\loader.py", line 66, in loadUi
    return self.parse(filename, resource_suffix)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\site-packages\PyQt5\uic\uiparser.py", line 1020, in parse
    document = parse(filename)
               ^^^^^^^^^^^^^^^
  File "C:\Users\PC\anaconda3\Lib\xml\etree\ElementTree.py", line 1218, in parse
    tree.parse(source, parser)
  File "C:\Users\PC\anaconda3\Lib\xml\etree\ElementTree.py", line 569, in parse
    source = open(source, "rb")
             ^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\PC\\anaconda3\\Lib\\site-packages\\linien_gui\\ui\\main_window.ui'`

deveshq, 2024-05-29:
I'm also facing similar issue. 

First, I tried to install the linien server directly using GUI. It's giving following error:

[image]

I also tried installing the server manually using the command `pip install linien-server` after connecting to red pitaya through SSH and got the following output:

```
root@rp-f06284:~# pip install linien-server
Collecting linien-server
  Could not fetch URL https://pypi.python.org/simple/linien-server/: There was a problem confirming the ssl certificate: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645) - skipping
  Could not find a version that satisfies the requirement linien-server (from versions: )
No matching distribution found for linien-server
```

I am using redpitaya STEM 125-14 v1.0 and OS version 1.04-18 and running the linien in a conda environment with python-3.10

bleykauf, 2024-05-29:
@felixhoff: You do not seem to have internet access on the RedPitaya. Make sure you do, e.g. by connecting it to a router (there are ways to give internet access while directly connecting the RP to your computer, but I do not have any exerience doing so). Some institutions also block internet access for RPs. As for the GUI issues, it looks like there are some incompatibilities regarding packages. Are you on Python 3.12 by any chance, see #401. Please try installing `linien-gui` in a clean conda environment with Python between 3.8 and 3.11.

@deveshq: Only RedPitaya OS 2.0 is supported for Linien 2.x, see installation instructions in the README.

bleykauf, 2024-05-29:
@felixhoff: Latest release 2.0.3 is also compatible with Python 3.12. Rest of my response still applies, though.

felixhoff, 2024-05-29:
For the GUI, I setup clean environment with python 3.8.19, I run pip install linien-gui then run the command linien and again I got the same error

```
(Cleancondaenv) PS C:\Users\fhoffet> linien
C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\paramiko\transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated and will be removed in a future release
  "class": algorithms.Blowfish,
Traceback (most recent call last):
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\Scripts\linien.exe\__main__.py", line 7, in <module>
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\linien_gui\app.py", line 116, in main
    app = LinienApp(sys.argv)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\linien_gui\app.py", line 48, in __init__
    self.main_window = MainWindow()
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\linien_gui\ui\main_window.py", line 87, in __init__
    uic.loadUi(UI_PATH / "main_window.ui", self)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\PyQt5\uic\__init__.py", line 241, in loadUi
    return DynamicUILoader(package).loadUi(uifile, baseinstance, resource_suffix)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\PyQt5\uic\Loader\loader.py", line 66, in loadUi
    return self.parse(filename, resource_suffix)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\site-packages\PyQt5\uic\uiparser.py", line 1020, in parse
    document = parse(filename)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\xml\etree\ElementTree.py", line 1202, in parse
    tree.parse(source, parser)
  File "C:\Users\fhoffet\Anaconda3\envs\Cleancondaenv\lib\xml\etree\ElementTree.py", line 584, in parse
    source = open(source, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\fhoffet\\Anaconda3\\envs\\Cleancondaenv\\Lib\\site-packages\\linien_gui\\ui\\main_window.ui'
```

bleykauf, 2024-05-30:
For some reason, Github Actions did not include the package data but when building locally on my machine it did. Anyway, fixed the underlying issue and also included the logo in the package. See new release v2.0.4.

closed 2024-05-30 by bleykauf

128164, 2024-06-20:
https://github.com/linien-org/linien/issues/405#issuecomment-2136889848

> I'm also facing similar issue.
> Only RedPitaya OS 2.0 is supported for Linien 2.x, see installation instructions in the README.

I'm having the same problem as him, currently linien is using version 2.0.4,redpitaya is the os version, I'm trying to upgrade it to 2.0 now. I'm trying to upgrade it to version 2.0. I'm testing linien version 1.0.1, using vscode, putty to make windows connect to redpitaya, and when I open this step "linien-server start", it keeps showing me that I need to enter the port [ PORT], and I can't change it. PORT], changing the port number to 18862 or 8080 or 22 is not correct, have you faced this problem or is it a problem with the way I am connecting to the test? This has been bothering me for days, looking forward to your reply!

Translated with DeepL.com (free version)

128164, 2024-06-20:
> [#405 (comment)](https://github.com/linien-org/linien/issues/405#issuecomment-2136889848)
> 
> > I'm also facing similar issue.
> > Only RedPitaya OS 2.0 is supported for Linien 2.x, see installation instructions in the README.
> 
> I'm having the same problem as him, currently linien is using version 2.0.4,redpitaya is the os version, I'm trying to upgrade it to 2.0 now. I'm trying to upgrade it to version 2.0. I'm testing linien 1.0.1 as a connection test, and I'm using vscode, putty to make windows connect to redpitaya, and when I open this step "linien-server start", it always shows me that I need to enter the port number [ PORT]. I have changed the port number to 18862 or 8080 or 22 but it is not correct, have you faced this problem or is there any problem in the way I am testing the connection? This issue has been bothering me for days, looking forward to your reply!

It's hard to say, I don't know how to upgrade redptaya os version 2.0, can someone please advise me, thanks a lot!

bleykauf, 2024-06-25:
https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html

PR #407 (2024-05-30): Release v2.0.4
Fixes #405