linien-org/linien #408: Offset voltage (Center) still on on FAST OUT 2 even when Sweep is paused

hmartinssonthorlabs, 2024-06-03:
When turning off the sweep by clicking on the Pause button in the Modulation, Sweep & Spectroscopy tab, the votage sweep is turned off but the offset voltage on FAST OUT 2 is still on until  0 is assigned as Center value.

It would be very good if the voltage on FAST OUT 2 is completely zero when the sweep is paused. 

Linien version: 2.0.4
Python version: 3.12.2
Red Pitaya OS: 2:00-35

bleykauf, 2024-06-03:
That is expected behavior. I would not see which other voltage should be output instead: If the user wants to sweep from let's say 500 mV to 800 mV, the output should not be 0 V if the sweep is paused in my opinion.

martinssonh, 2024-06-04:
I get your point. I think it's me who's not using Linien as intended. I'll rest my case :)

closed 2024-06-04 by martinssonh