linien-org/linien #414: Not possible to change PSD algorithm in noise analysis

martinssonh, 2024-07-17:
### It's not possible to change the PSD algorithm in the noise analysis.
When trying to change the algorithm in the pull-down menue from Scipy to LPSD, it always defaults back to Scipy. However, I'm pretty sure it still uses LPSD (the measurement taks longer time and contains just a few glitches) despite what the menu says.

Also, the "psd_algorithm" setting is missing in the devices.json file.

By adding the entry "psd_algorithm": 0 last in the devices.json file, I can force the linien server to use Scipy (Welch algorithm). The PSD spectra measurement is now must faster and contains several glitches.
Also, the DEBUG entry
"linien_gui.threads             INFO     Parameter psd_algorithm differs: local=0, remote=1"
informs that the PSD algorithm is changed from 0 (WELCH) to 1 (LPSD).

Since users most likely will use the LPSD algorithm over the Scipy (Welch) algorithm anyways, due to all the glitches, this is maybe more of an annoyance than a bug.

Linien version: 2.0.4
Device: STEMlab 125-14 Low noise
OS: 2.00-35 beta

bleykauf, 2024-08-13:
Thank you for reporting this. Attempting to fix this in #421.

bleykauf, 2024-08-14:
Found the issue: I did mix up the numbering for the two algorithms in `psd_window.py`. Now simply casting the `PSDAlgorithm` Enum to a list (https://github.com/linien-org/linien/pull/421/commits/b59521f7a1c9fffb01fbe7be886b16f80ae190d3). Otherwise declaring a Enum does not actually make any sense anyway...

Also made the paramter restorable.

closed 2024-08-15 by bleykauf