linien-org/linien #434: Unable to run linien-server - flashing FPGA

yonicon, 2024-12-05:
Hello, 
My RedPitaya is STEMlab 125-14 Z7020 
I flashed the RP with a new OS ver. 2.04 build 35 (RedPitaya_OS_2.04-35_stable.img.zip)
I'm trying to use linien version 2.1.0 (linien-2.1.0.exe) 
I installed the linien-server (version 2.1) on the RP using pip. 
When trying to run the linien-server I get this message and the SSH connection crashes immediately after - 
[image]

I checked sha256sum of gateware.bin and it is ok. 

This seems very much like the problem in 
https://github.com/linien-org/linien/issues/394

I'd appreciate any help, 
Many thanks 
Yoni.

Lightsaver7, 2024-12-11:
This should be related to the difference in the FPGA image used by STEMlab 125-14 Z7020 (which uses the Zynq 7020 SoC) and a "normal" STEMlab 125-14, which uses Zynq 7010 SoC FPGA. Most likely the FPGA pin definitions would need to be changed in the .xdc file (or a logic added that recognizes the specific board model Linien is running on).

ZHT-wed, 2026-10-31:
Hi @Geosemn, thanks for working on this. I'm trying to get Linien 2.1.0 running on a STEMlab 125-14 Z7020 LN (hw_rev=STEM_125-14_Z7020_LN_v1.1, Red Pitaya OS 3.00-57). The released gateware.bin fails to configure the PL on this board — fpgautil -b .../linien_server/gateware.bin reports "Error after writing image data to FPGA" and fpga_manager ends in state 0xffffff92, which looks like the same problem as issue #434 (the shipped bitstream targets xc7z010-clg400-1).

Do you have a compiled gateware.bin from this branch that you can attach here (or link as a release asset)? I'd be glad to test it on the Z7020 LN board and report back with the loading logs. If you'd rather not share a binary, the exact Vivado version and build steps you used would also help me reproduce it.

Geosemn, 2026-11-10:
> Hi [@Geosemn](https://github.com/Geosemn), thanks for working on this. I'm trying to get Linien 2.1.0 running on a STEMlab 125-14 Z7020 LN (hw_rev=STEM_125-14_Z7020_LN_v1.1, Red Pitaya OS 3.00-57). [...]

I have stopped working on this it's been a while. I have both a gateware compiled, vivado project and an SD card copy that's actually working on the Z7020 with autolock. pm georges.semaan@ulb.be for more info, I cannot attach everything here.